marchingband/wvr #2: Feature Request: Sample-Accurate Looping

TheBatmosphere, 2021-07-24:
I'd like to be able to hold a note and have it loop seamlessly. In 1.0.1, there's a gap.

BBGuy, 2021-11-05:
> please allow start index, loop index, loop end index, and end index per sample. this will allow many amazing sound possibilities. it would be great if you could feed WVR files with these loop points defined and saved in the header.
> 
> then of course all the loop types like forward, backward, alternating.
> 
> if you really want to go the distance, to allow the sample start, loop start and loop end points to be controlled with cc... that allows some wizardry and i would buy beers for you.

That's a better description and more flexible than my feature request in https://github.com/marchingband/wvr/issues/7

This would be amazing

solipsvs, 2021-11-06:
in the old school tracker interface was included a way to address your loaded samples directly and with a sample start index 0 to 126 or some other commonly used number system.  this would allow you to slice the audio exactly half way or 3/4 or whatever the resolution allowed for, for every note.  for every sample.  so you could apply this to anything loaded in the sample pool.  of course, it only really makes sense if your samples are formatted to the process of subdivision, like drum beats.  it also enabled some rudimentary effects like echoey time stretch etc.  so a way to access those points in a sample easily is definitely conducive to creative use.

marchingband, 2021-11-13:
@TheBatmosphere FYI I found a bug that was causing a long pause between loops. With that bug patched, there is still a very small gap (it's not sample accurate) but its much better. Worth trying out, it's fixed in 1.0.5 and after. I will work on this soon though :)

marchingband, 2021-11-19:
@TheBatmosphere @solipsvs With this ADSR feature, does the playback jump immediately to the release index, when a note-off event occurs? Or would it be acceptable for a note-off to simply stop the looping.
For example, if a note-off happens while playback is mid-way through the looped section, can the playback continue from where it is, finish the looped section, and then proceed past the loop point, and play the release? Or does it need to skip the remainder of the looped section, and immediately play the release section? I am assuming the latter is the expected behaviour, but just want to confirm.

marchingband, 2021-11-19:
I am close to having sample accurate looping working now, but, the loop does not x-fade from the end to the beginning, it is just a hard loop.
This means that your samples will have to be crafted to loop well without a 'pop'.
Is this acceptable?
(note that this is a separate feature from ADSR. This is just selecting 'loop' from the 'playback mode' dropdown. In this mode the whole sample will just loop)

solipsvs, 2021-11-20:
setting loops precisely is acceptable to me

TheBatmosphere, 2021-11-20:
Yep. I just wanted WVR to play loops as intended. Thanks!

PR #34 (2022-02-01): Looping
resolves #2
Audio engine now hold the first read from eMMC in a PSRAM buffer, to allow seamless sample-accurate looping.
closed 2022-02-01 by marchingband