marchingband/wvr #4: Feature Request: Load Multiple Samples at Once

TheBatmosphere, 2021-07-24:
Loading one file at a time is inefficient. I'd like to be able to select multiple files and have WVR load them into sequential (or designated) open slots. One way to do that would be to prefix the filenames with the MIDI note number. (This is how the SparkFun Tsunami works.)

BBGuy, 2021-08-01:
Thumbs up for this one. Especially important if you are using a multi-sampled instrument with one sample/pitch for each key.  

Another improvement can be supporting racks to allow for multiple samples per each key. This will probably need a naming convention something like:
{channel}_{note number}_{rack layer}_{filename}
This will also allow creating complete setups on your computer and loading them at once.

solipsvs, 2021-11-02:
+1

essential!

marchingband, 2021-11-04:
I have proposed a solution to uploading multiple files at a time, which works for ranges of notes, or ranges of breakpoints within a rack. The details are in issue #8. Now I am working on a solution to bulk upload directories of racks, so a whole virtual instrument could be uploaded from a well formatted folder.
What I have working is : 
select a range of notes using shift-click and alt-click, and then shift-click the "select file" button
A file dialogue opens (if you are using Chrome, and maybe firefox) that allows the user to select a directory.
This directory should contain several subdirectories, each of which will correspond to the notes in the selected range. Each subdirectory should contain all the audio files for the racks for a note.
Both the subdirectories, and the files they hold, will be sorted alphabetically, and placed into the appropriate racks sequentially, using the default break points.

I know this is a little hard to imagine, I will be finished the PR soon, and folks can give it a try :)

TheBatmosphere, 2021-11-06:
This works great in 1.0.4! I loaded 35 files with one click, then changed their parameters with a second click. Thanks!

closed 2021-11-13 by marchingband