marchingband/wvr #6: Noise on output

BBGuy, 2021-08-01:
Hi, 
Really excited about this, looks like a great design. 
I just had my first play with the board and got it to work using your instructions but I am getting a lot of noise on the audio output.
I am using it with the USB Backpack and the wvr_usb firmware.
I am only powering the mane WVR board using the pins for now.
I am using a USB power pack 
All connections are on a breadboard:
mini stereo jack connected to L, R and one of the GRD pins connected to a minirigs speaker.

I am getting two types of noise that sound together like a metronome:
1) A repeated higher pitched click
2) A slower lower repeatable click.
 the first goes away if I turn WiFi off using one of the pins but I could not find a way to get read of the second.

Maybe I am doing something wrong. I got two more boards and can try and solder pins to the KVE only if you think that will make it easier to test. Am I correct in understanding that the USB backpack is off unless powered directly?

Hope you can help?

marchingband, 2021-08-01:
Thanks for writting BBGuy! And thanks for your other feature requests :) they are awesome.

The USB backpack is always on when WVR is on, they share the 3.3v rail.

With the WIFI off, there should not be any noise. 
Having WIFI on does result in some noise. Here is how I describe the level of this noise : If the volume on your speakers is set to a reasonable listening volume, and you are using a good loud sample, this noise is just barely audible. It appears as a rhythmic click, and some data noise when WIFI is working on a transfer. The WIFI radio uses a lot of power, and I could not find a way around this noise. I will continue to work to reduce this noise on WVR V2. Luckily, turning off the WIFI eliminates this noise completely.

The most likely source of external noise is the power supply. I recommend experimenting with other power supplies as a first step.
Failing this, you could try another of your boards. There is always a chance that something was damaged in shipping, etc.

BBGuy, 2021-08-01:
Thanks, @marchingband for the quick reply.
I will keep experimenting and report back.
I don't mind some noise when the WiFi is on as I expect to only be using it to configure the board

BBGuy, 2021-08-03:
I have tested with the WVR only (no USB Backpack) and was not getting the clicking noise, with or without WiFi.
I have ordered some stackable headers from China and when I get them (2-3 weeks) I will test with the USB Backpack. If I can find a way of not permanently connecting the backpack in the meantime will update sooner.

marchingband, 2021-08-03:
Are you able to share a photo of your soldering? I have a strong suspicion there is a physical problem. Perhaps you got the orientation backward?

BBGuy, 2021-08-03:
Let me know if you need specific detail shots.

[image]

marchingband, 2021-08-04:
That looks ok from what I can see.

You can add pins to wvr, and the USB  backpack separately. Place both on a
breadboard, and connect 3.3v, 5v, and “USB midi” pins between the boards,
and it will work the same. This may help you debug. I think the most likely
thing is something got shorted in your soldering process. Have you tested
all the adjacent pins for shorts?

BBGuy, 2021-08-05:
Thanks, I did test for shorts but will double-check and will try your suggestion, it makes sense.

BBGuy, 2021-08-08:
Looks like I got to the bottom of my issue.
One of my ground pins was not connecting to the KVR. It was the one next to the 3.3V pin, maybe it's connected to a large ground plane as it did not easily flow. I ended up putting flux on the area and managed to get the solder to flow through.  
The little gaps on the side of the KVR are really helpful for checking continuity. It may be an idea to add continuity checks for all the pins in the instructions as well as checking for shorts between the pins.
Have not connected anything to the backpack but will do some more testing and report back.

marchingband, 2021-08-08:
@BBGuy thats great! I have updated the instructions :)

BBGuy, 2021-08-09:
Excellent, really clear instructions. 
Closing the issue as it is fixed. will reopen or create a new one if any more noise-related issues come up. 
Thanks again

closed 2021-08-09 by BBGuy

marchingband, 2021-08-10:
@BBGuy great! glad to hear. Thank you!