marchingband/wvr #8: Feature Request: Copy Settings to Other Notes

TheBatmosphere, 2021-08-21:
Setting up a group of notes to play (for example) a Mellotron or drum kit takes a _lot_ of clicks. I wish I could shift-click notes and change the same parameters at once, or copy and paste all parameters for one note onto another.

marchingband, 2021-08-21:
Shift-click to select a group is a great idea and shouldn’t be hard to
build.
I will implement this ASAP. Almost out of the weeds with soldering :))))
will be adding features soon :))))
Thank you!

TheBatmosphere, 2021-08-22:
Yesss! I just ordered another WVR. Looking forward to the next project.

marchingband, 2021-11-03:
I am setting up shift + click to select a range, and alt/option + click to add/subtract an individual note from the range.
Any changes (including file selection) made in the note details pane will be applied to all the selected notes.

If the user clicks the "select files" button while a range is selected, then it is possible to select multiple files, which will be distributed across the selected notes, according to the order they appear in the finder window. If only one file is selected in this scenario, then that file will be assigned to all the selected notes.

I am also setting shift + alt/option + click to set the "pitch interpolation target", which means the note which is at pitch 0 when interpolating pitch across a range of selected notes, the interpolation of the other notes is determined by their relationship to this selected note. If an interpolation target is set, then clicking the "select file" button will trigger a pitch interpolation action.

Clicking the "select file" button while a range is selected will prompt the user to confirm the action, to avoid accidents.

This update should also cover issue #4, and issue #10

Let me know if this sounds good!

TheBatmosphere, 2021-11-04:
Thanks! A few thoughts:
1. I'm not clear on the pitch interpolation concept; can you give an example with note names? 
2. Would it be simpler to shift-click once to add a note to the selection and shift-click again to subtract? Other software works that way. 
3. A comment on #5 requested more contrast on the selected item. That would be helpful. 
4. Let me know if I should post this as a separate feature request, but it would also be helpful to be able to drag samples from note to note. I typically wish for that feature after loading samples into the wrong slot or changing my mind.

marchingband, 2021-11-04:
thanks for the notes @TheBatmosphere :)

1: If I select the range 40-43, and select 42 as the interpolation target, then the pitch shift will be assigned like :
40 -> -2
41 -> -1
42 -> 0
43 -> +1

2: Using shift-key for range selection and alt-key for individual toggle, closely matches how file selection works in the OSX finder, and is pretty powerful. Let's try it on for size first.

3: thanks for the reminder I'll do that in this PR as well :)

4: this could be done, but would take some work, because copying a note which has previously been uploaded to WVR is something that would have to happen on the WVR itself (the browser no longer has access to the file, or it may not even exist on the computer being used). I will have to build a new server endpoint, and figure out how to do that. If after these updates it still feels important, yes, please open a new issue :)

closed 2021-11-13 by marchingband