marchingband/wvr #9: Support request: power the board directly from 5V?

BBGuy, 2021-09-04:
Is it possible to power the WVR board directly from 5V without using the USB connection? (i.e. connect using the pins)

marchingband, 2021-09-04:
Yes both the 5v and 3.3v pins can be inputs or outputs.

DennisSchm, 2021-09-17:
Hey! I would like to join in on this question. When I power my board over micro USB everything works just fine but when I connect GND to ground and 5v to + nothing happens. 

My understanding is that the 5v pin can be used as output if the board is powerd over micro USB and is automatically noticing if it's powered over the 5v pin and uses it as input. Am I wrong in this asumption?

marchingband, 2021-09-17:
No there is no sensing or anything like that. Applying power to the 5v pin
should power the board.

closed 2021-11-03 by marchingband