marchingband/wvr #12: Possible cause of gap in loop - toPcmFx

jazzbiddy, 2021-11-01:
Hi @marchingband,

After you helped get me back up and running after issue https://github.com/marchingband/wvr/issues/11, I poked around a bit,  taking into account what you had mentioned...

> In wav_player.c, the struct buf_t needs to have a special buffer to store the first chunk, maybe called buffer_head or something. Then the engine needs to detect the end of the last chunk, and switch over to the buffer_head, reset all the pointers and stuff, and then proceed as normal.

I put a bunch of debug statements in several spots and stepped through things. While I get what you're saying above, everything was telling me that the code in wav_player was working properly - that what was happening wasn't just that the wav had passed the end of it's buffer and was just not being detected. So after some digging, here's my guess at the issue....

Additional "dead space" is being added to the file by the web UI prior to being uploaded to the WVR. The culprit seems to be the toPcmFx method. In fact there was a commented out call to `toPCM` sitting there for me to try - and when I used that, the correctly sized file was uploaded. Here are some screenshots of debug prints , but you probably know what I'm talking about.

The screenshots are all for the same single wav file, but I've tested with many and they all loop seamlessly using `toPcm` 

### Here is the size of the original file.
[image]

### I put some serial prints in server.ccp>>handleWav(..)
[image]

### When I sync'd the wav file, I see `total` value for the file size is bigger than the original
[image]

### I also printed out what the buffer size was in wav_player.cpp - this made it to the web console, via websockets
[image]

[image]

### I then logged the UI's sync.js uploadWavs function
[image]

### I could see the 2 sizes
[image]

_NOTE: The size is slightly different than the original. I'm not sure if that's the wav header being added/removed. That doesn't seem to impact playback_

### I modified the sync.js to upload the output from `toPcm` instead of `toPcmFx`. Now when I played the file it looped perfectly without any gaps!! And the buffer size printed the correct size value
[image]

[image]

Here's a before/after demonstration

### Before - using `toPcmFx`

https://user-images.githubusercontent.com/17692134/139611552-bd25674a-75a9-4bc0-98d6-c31f4dd7e15d.mp4

### After - using `toPcm`

https://user-images.githubusercontent.com/17692134/139611578-95eefbe7-58fb-4be6-a7cb-f9e7f06fa0be.mp4

So what does it all mean??? If this indeed the root cause of the gap, then maybe a short-term solution is to add a toggle so users can enable/disable the effects section.  If I knew react better I'd add that in and maybe post a PR. But the proper fix is probably to see what's going on in `toPcmFx` that might be adding the gap, and try to correct it there. 

Hope this helps. While I was messing around, I did notice some strange behaviors that I'll write up once I make sure they weren't something I introduced while debugging.

marchingband, 2021-11-01:
@jazzbiddy amazing!
I think `toPcmFX` is adding a reverb tail, even when the reverb volume is 0 (reverb not used).
That is an amazing bug to catch, thank you!
Even with this bug removed, I don't think we will have achieved "sample accurate looping" as requested in issue #2.
512 bytes (the size of the audio buffers) goes by very fast (I think about 3ms), and is not easily perceptible, except under very specific circumstances.
Each buffer is initially zeroed, and then the available data is copied in. If the last chunk is less then 512 bytes, the zeros remain as padding, and the whole buffer is summed into the output. A loping sample will be restarted, and resume playing at the next pass, but the padding in the buffer has already been written to the output.
I will fix up the bug you you found, and make a PR :)
It's really amazing to have another pair of eyes on the code, so thank you so so much :))))

closed 2021-11-01 by marchingband

marchingband, 2021-11-01:
Ok I've merged in a fix, if you are using git, you can `git pull` or if you downloaded the .zip file from releases, you can download the new version v1.0.3, and replace the version you have in the /libraries folder :)
Let me know if you encounter any issues with it :)

commit 0a0e8751 (2021-11-01): fix #12
commit f44b99c9 (2021-11-01): Merge pull request #13 from marchingband/issue12

fix #12