marchingband/wvr #16: Feature Request:  Round Robin

solipsvs, 2021-11-02:
allow cyclic and random playing of similar files (in a folder maybe?) to increase realism and organic sound for snare drums for instance but for other uses too!  allow for every velocity layer and every midi note number,

TheBatmosphere, 2021-11-04:
That would be wild! Maybe it could be implemented in the editor by adding a drop-down drawer that would hold additional samples uploaded to that note.

[image]

solipsvs, 2021-11-04:
indeed!  this would aid realism from the start

marchingband, 2022-10-18:
This could be implemented by consuming voices. Each note has 2 extra drop-downs, 1 for the %, one for the number of members in the round-robin. The other members of the round-robin are placed in the next consecutive voices, in the same voice/rack slot, where the % of all members must total 100%

Gnuv, 2022-10-24:
Now that we have a parser on the way for sfz format, let me suggest a compatible solution for round-robin :

The idea is to use 16 <group> headers to identify the 16 voices.
Then, to open the possibility to set the program change of each voices with the opcodes loprog= and hiprog= https://sfzformat.com/opcodes/hiprog

Then, to use the opcodes lorand= and hirand= https://sfzformat.com/opcodes/hirand to set the round-robin :

<group> //Voice 01
loprog=1 hiprog=1
<region> key=72 hirand=0.3 sample=kick_rr1.wav

<group> //Voice 02
loprog=1 hiprog=1
<region> key=72 lorand=0.3 hirand=0.6 sample=kick_rr2.wav

<group> //Voice 03
loprog=1 hiprog=1
<region> key=72 lorand=0.6 sample=kick_rr3.wav

With this solution, we can even have multilayers sounds, each one with his own round-robin (for example an attack sound with two round-robin on voices 1 and 2 and a release sound with three round-robin on voices 3, 4 and 5).

Of course, this could work only if several voices can be played together in you code. If not, a possibility is to put the lorand= and hirand= opcodes on the group and having the same round-robin for all the voices.

TheBatmosphere, 2023-09-30:
How about adding a checkbox to racks to make samples switch sequentially rather than by velocity?
[image]

marchingband, 2023-10-01:
That is a tidy idea for sure, I did considered it, but don't we want round-robin AND velocity layers ideally?

TheBatmosphere, 2023-10-13:
I did mention having both 😁, but now think round-robin combined with standard velocity-to-volume would be easier to use. In other words, selecting this box would cause each new Note On event to trigger the next sample in the group and velocity would affect any given sample equally. (Or not at all, if you set fixed velocity response.)

solipsvs, 2023-10-14:
lost my password and bricked my device.  is there a way to get another device?  sorry to bother you all

marchingband, 2023-10-14:
You have 2 choices for a new WVR
https://www.sparkfun.com/products/21307

https://www.tindie.com/products/ultrapalace/wvr/

but if you get an FTDI module to communicate with the WVR (or if you own one already) you can use that to reset the memory, including the password.

marchingband, 2023-10-14:
The .sfz format has round-robins for each velocity layer so I would like to follow that standard.

TheBatmosphere, 2023-10-23:
> The .sfz format has round-robins for each velocity layer so I would like to follow that standard.

Even better!

Sporradik, 2024-05-22:
I just bought one of these as recommended by a friend with the hope of getting this working more or as described here. 

Is there any way with the current implementation to have a single input trigger multiple samples (either at random or in order?)

marchingband, 2024-05-22:
Hi @Sporradik are you Steve's friend?
There is not a way to do that.
You would have to write a little Arduino code. I would be happy to help if you need it.

marchingband, 2024-05-22:
@Sporradik I am assuming when you say trigger that you mean using the GPIO. If you are using MIDI, there are many way to do it, let me know if you are.

Sporradik, 2024-05-25:
@marchingband Yes I am Steve's friend and yes GPIO. Thank you for offering! Steve also offered to help me out by writing some code to generate a random MIDI note. Do you have any suggestions on the easiest way to do this? Is it possible to do this without extra hardware? I am not very familiar with the world of hardware programming.

marchingband, 2024-05-28:
It will be streighforward to do this, and you are in good hands with Steve, no extra hardware needed, just a few lines of code. HMU anytime for support should you need it though! Steve has my number, and also feel free to join the discord https://discord.gg/ZfXuBkzec8