marchingband/wvr #20: Feature Request: Clear Sample

jazzbiddy, 2021-11-05:
I'm not sure if I'm missing this somewhere, but it would be great to have a way to remove a sample from a note after you've assigned one to it.

marchingband, 2021-11-05:
This is an embarrassingly huge oversight!
I will add this ASAP.
From a UI perspective, would it make sense to hit `delete` on the keyboard, to erase the currently selected wav(s) ?

jazzbiddy, 2021-11-05:
oh man... great question. Maybe a button added to the right side set of buttons as well as the keyboard key - but you'd need to have a confirm dialog for both, because, you know, accidents happen and it would suck to mistakenly delete the sound. 

[image]

of course, when cleared, and a sync was performed, the note wouldn't play anything anymore.

marchingband, 2021-12-18:
This is implemented in v1.0.9

closed 2021-12-18 by marchingband