marchingband/wvr #21: Possible Issue - Updating Sample to one with a smaller length leaves some data from previous sample

jazzbiddy, 2021-11-05:
I'm not sure yet if this is an issue, but I wanted to log it in case anyone else has experienced it. 

What happened was I uploaded a sample to a note - say a guitar riff that was 20 seconds. Then later I uploaded a new sample to that same note, say a horn hit that was 5 seconds. When I played the note, I heard the 5 seconds of the horn hit, but then the guitar riff that I had uploaded previously will play for the remainder of it's length, probably 15 sec. 

Its as if the data that a file occupies in the WVRs memory for that note is not fully erased when a new file is uploaded to the same note. 

Like I said, I still need to check that this is 100% reproduceable.

marchingband, 2021-11-05:
This is a very strange bug! I havn't experienced it, so I am assuming it occurs under some very specific circumstances. Let me know if you are able to reproduce it and how!

marchingband, 2021-12-18:
I believe this bug was caused, of all things, by long file names corrupting the `length` property in the wav_file_t struct.
v1.0.9 fixes the filename bug, and I am hoping this also fixes this bug. @jazzbiddy If you update to the new firmware, the next time you sync, it should trim and \0 terminate all the filenames correctly. If you are still able to reproduce this bug, please let me know.

PR #36 (2022-02-17): Asr looping
overhaul the audio engine for much higher speed. There is now some extra processor time to play with, to implement more midi controls like panning, expression, etc, in a future PR :)
add ASR-looping to the engine and the WEB UI
add MIDI CHANNEL to WEB UI, and to midi engine : WVR can listen on only one midi channel, and is not stuck in OMNI mode :)
Each midi channel now maps to its respective voice at boot. ie. midi events on channel 5 play sounds from voice 5, etc. Program Change events still modify that routing, but they reset to defaults at boot.
upgrade to ESP32-ARDUINO v2.1
fi [... cut]
closed 2022-02-17 by marchingband