marchingband/wvr #22: Feature Request:  Wavetable Playback mode

solipsvs, 2021-11-05:
this is a little scientific but with special looping and double the polyphony usage, wavetables can be played.  wavetables are amazing synth tools that are easy to get sophisticated sounds out of with just sample payback capability.  i can explain more if anyone needs more info.

marchingband, 2021-11-05:
So, I think there is a cool solution for this feature. WVR has a ton of extra RAM sitting around, that I knew would come in handy eventually. If the user uploads a sample in `wavetable` mode, when WVR first boots, it could load the entire sample into RAM, meaning that it would bypass the whole audio engine, and it would be easy to implement any kind of re-ordering of the playback. Playback could jump around, reverse, loop from any point, without any overhead with reading from eMMC in advance.
The drawback to this is that there is a fixed amount of RAM available (about 14 MB or so, in 4MB chunks) so there is a hard limit on the size, and number, of samples that could be set up for `wavetable` mode at a time.
If the user switches voices (banks), then WVR could take a few milliseconds and load new samples into RAM, but each voice would have a limit.

If you want to get deep into what features a `wavetable` mode would offer, I am all ears, I don't have much knowledge in this area. You could link some articles too if you like. I will probably set up a whole new screen in the WEB UI for this purpose, so lets get everything in that we want.

marchingband, 2021-11-05:
If possible, lets avoid anything that needs LFO, etc, for now, and focus on features that just require manipulating the order in which the samples play back. that could include mixing to make the transitions smooth, but, currently WVR has no concept of an LFO or oscillator of any kind.

solipsvs, 2021-11-05:
that sounds very cool.  so i am using WVR as a supplimental device so its ok with me if it has limits.  everything has limits.  i like to be able to use my machine's entire capability to make one note as in unison mode so also if any mode i use reduces cpu headroom, ill just use it minimally.

the term wavetable is relative.  to some its just one waveform or sample bed to be used as an element in a finished patch, sort of like an oscillator.  to others, its a bit more involved:  its a collections of frames that represent a sound but with a timbral change spread across all the frames, which are essentially all single cycles.  each frame has a unique character among all in the wavetable.  so the system plays one cycle looped and cues up the next one ready to crossfade to it and on to the next frame.  you manipulate the frame index and your system dynamically fades between frames.  this allows animation and character of any kind, since its only sample data youre playing back.  so your wavetables can have any character and retain the systhesis element that can be repurposed in your composition.  

ance we get wavetable playback, what you do between multiple wavetable oscillators or indeed manipulation of the frame becomes the next level of sound shaping.  thats where FM and other DSP transforms it to another level.

usually a wavetable system will have a defined file type with restricted number of frames, standard frame size, etc.  wavetables are only good if formatted precisely.  but the sky is the limit.  what really comes into play with a good wavetable oscillator is the quality of the pitching algorithm.  crisp interpolation results in much more desireable effect if you ask me.  i cant stand tinny waveforms.  thin is your enemy!  of course there is a threshold.  many poly instruments are a little thin so the notes fit and arent too big!

anyway, once you have a waveform output, theres much modulation to do!  so envelopes are a definite must, LFOs with BPM sync, filters, fx...  all after DSP is a reality.

solipsvs, 2021-11-05:
yeah its a little advanced.  i thought it might be too advanced.  ok later on this one hopefully lol