marchingband/wvr #23: Feature Request: Adjustable Fadeout to Prevent Note-Off Clicks

TheBatmosphere, 2021-11-06:
WVR sounds cut off abruptly on Note-Off. The SparkFun Tsunami has a thoughtful feature that lets you set a default release time of 0–2000ms and override that live by sending a MIDI CC. (The Tsunami uses CC1 and CC74.) I set the global release to 5ms, which prevents clicks.

[Tsunami MIDI implementation](https://robertsonics.com/tsunami-user-guide/#midi-implementation)

marchingband, 2021-11-06:
The latest version of the firmware does fade out samples on note-off, but very quickly (about 2ms). If you are using the stock firmware, it may not have that feature. v1.0.1 and v1.0.4 should not click.
An adjustable release time could certainly be achieved, I just hadn't thought of that!
Would a global control be good enough, or would a note, or sample specific release be useful as well?

TheBatmosphere, 2021-11-06:
Thanks for asking! A modulatable volume envelope would be the ideal solution, but for my playing, the Tsunami’s global release setting and CC override have been fine. (I’m using it to trigger sound bites and loops rather than do fancy synthesis; I have other instruments for that. I expect I’ll use WVR that way too, though I’m loving the sample-scanning and looping suggestions others are making here.)

CC override example: Raising the mod wheel increases the release time.

marchingband, 2022-02-28:
implemented in v3.0.0
Sending CC 72 with a value of 127 will cause around 30ms release time. Sending CC 72 with value 0 will set the default release time, about 1 ms. The release times scale linearly between the 2 with all the intermediary values.

closed 2022-02-28 by marchingband