marchingband/wvr #24: USB Sound Response Lags on Repeated Notes

TheBatmosphere, 2021-11-06:
When triggered from basic MIDI, WVR retriggers notes as fast as they're hit. (Although repeated notes stack up and distort, which may be another issue.)

When triggered from USB, the sounds start to lag, as if the triggers were filling a bucket and then slowly trickling out.

Play a snare roll to hear this. Play a cymbal roll to hear the stack-and-distort issue. That happens with both MIDI and USB input.

marchingband, 2021-11-06:
The stacking can be avoided by setting `retrigger-mode` to `restart` for that note, in the WEB UI. The sample will fade out ASAP before the next one is started.

The lag is not something I had noticed :( It's almost certainly in the USB Backpack firmware. Unfortunately the usb-host library I use was never completed, so I had to hack together something that "worked". I'm not surprised there is a bug in there. Not being very familiar with USB protocol, which is incredibly dense, it will take some mucking around to fix this I expect. thanks for pointing it out, and I'll get on fixing it ASAP!

marchingband, 2021-11-06:
Also, can you confirm that you have updated the firmware on your USB Backpack?

marchingband, 2021-11-06:
In your screencaps in the google group, the USB firmware seems to be the wrong size, so it may have silently failed when you went to update it.

TheBatmosphere, 2021-11-07:
Ah! Setting retrigger mode to **restart** and updating the USB firmware fixed the lag. (What is the use case for stacking a single note?) I can still get notes to distort by playing fast rolls.

marchingband, 2021-11-07:
That's great!
Ok I simplified the procedure a bit. During a re-trigger, the old sample is faded (over about 2-3 ms) but the new sample is started immediately, so long as there is a free buffer for it.
So, if you are triggering very loud sounds very very rapidly, that could create a problem.
The easiest solution is to lower the volume of the sample (which you can do in the UI).
WVR doesn't really set up any headroom for the output. If it mixes 2 wavs, they both go through at full volume, and if they add up to more then 0db, then there will be clipping. I did this on purpose because it's nice to have control over that stuff, and I trust users to manage all the headroom themselves.

marchingband, 2021-11-09:
@TheBatmosphere your experience of distortion led me to realize there was a bug in v1.0.4.
It's likely that this was not distortion, but glitching in the audio engine, it sounds similar to distortion, but it's actually the sounds of a buffer underrun.
v1.0.5 fixes this, and I am getting 18 voices of seamless polyphony once again :)

TheBatmosphere, 2021-11-09:
Nice! Yes, it was more of a choppy sound than fuzz. Can’t wait to try the new version.

closed 2021-11-13 by marchingband