marchingband/wvr #25: Feature Request:  Mute groups, excplusive groups

solipsvs, 2021-11-06:
i want my hihats to cut each other off and i want to make a bass patch play 1 note guaranteed.  we jsut need a way to do a quick fade on the note thats going away so theres less of a click?  im not sure how some samplers are able to do that without clicks.

marchingband, 2021-11-06:
WVR has this feature for a single note: re-triggers can be set to end currently playing samples for that note. This is set on a note-for-note basis.
There isn't a way to create exclusive groups yet, I will add that for sure, as well as mute groups.
It only takes 2 ms to cleanly fade out a sample, so it's achieved on WVR by adding that small bit of latency to the newly triggered note. The old one fades ASAP and then the new one starts ASAP.

solipsvs, 2021-11-08:
that latency is clever and 2ms aint much!

marchingband, 2022-02-18:
@solipsvs looking for more info here.
Do the newer firmwares fulfil this need? Within a given rack note, if in "retrigger" mode, notes will but each other off.
I could try to implement this same thing BETWEEN notes ... but I'm not sure what the use case would be

solipsvs, 2022-02-18:
so samples assigned to different note values could cut each other off?  its a useful thing for musicians who play from a purely sample based kit only.  i would definitely use this feature.  im sorry. i havent tried any new firmware yet.

marchingband, 2022-02-18:
ok cool, so that sounds like it would be called an 'exclusive group"?
What would a mute group be? I dont have any concept of muting a note, or voice, currently.

solipsvs, 2022-02-19:
a mute group is the exact same thing, just a different name foir exclusive group.  you would assign each sound to any of many groups where they all will cut each other off when another sound in the group is triggered.  so like when a closed hi hat sounds within the duration of an open hi hat.  its used to mimic the natural behavior of closing the halves of the hi hat.  you could also maybe hit a crash cymbal and mute it with your hand to get a gated effect.  you could also make several bass sounds in your soft synth or wherever you make synth sounds and write a bassline where the individual samples cut each other off, maintaining a mono output, as is the norm for basslines.  or maybe you could use a reversed reverb tail of a clap sound and make the forward clap sample cut it off, making a dramatic lead in up to the clap.  there are many ways this could be creatively used.

marchingband, 2022-02-19:
Ok cool, thanks for clearing that up!
This will be easy to implement in the engine, but presents a UI challenge, to fit into the current interface.
I'll prioritize this feature though :) it does sound very useful. I actually manually wrote this functionality into the firmware for the Thames pedal : https://github.com/marchingband/wvr/blob/main/examples/wvr_thames/wvr_thames.ino#L142