marchingband/wvr #28: Use a name-0value pairs file format to specify sample metadata

BBGuy, 2021-11-08:
We had the idea of using the file name before. Using name-value pairs will give us more flexibility to add new options. 

This will have two separators: 
1) **Name**-**Value** separator for each of the parameters
2) **parameter**_**parameter** separator

Example:
name-Kick1_lstart-50_lend-1000

* Name: Kick1 
* Loop start: 50
* Loop end 1000

TheBatmosphere, 2021-11-09:
This sounds promising. What is the function of the pipe character in the example? I find it hard to read. Maybe use a plus sign or ampersand to separate pairs? I'm thinking of URL query strings.

marchingband, 2021-11-13:
There are a few problems with file name conventions like this.
First off, in order to modify the loop points, the user would need to open a finder, rename the file, then re-upload it to WVR. With a large file this could be a very tedious process, if there were lots of fine adjustments to make.
 
Second, it creates the possibility of mysterious problems if, for example, someone uploads a file with "_lstart" anywhere in it, and is mystified by the results.

From a coding perspective, I would much rather write some new UI code, then build a whole string parser, so my preference is a UI solution. Is there a need for the file name convention that I'm not seeing? If not I will proceed with a UI solution.

marchingband, 2021-11-13:
FYI this feature will require an eMMC reset, since it adds some members to the `struct wav_file_t`, and the file layout in eMMC will have to change. so it will result in a major version bump, up to v2.0.0

marchingband, 2021-11-13:
"require an eMMC reset" means that all the samples on your WVR will be erased when you do the upgrade.
There may be a way to avoid this in the future, which I have been considering (storing all the wav LUTs in flash instead of on the eMMC) so this may be the only time this ever has to happen. That change may take a lot of work though, so for now I will just keep with the eMMC reset approach when the wav file structs change.

marchingband, 2021-11-13:
unfortunately using the "backup/restore" feature on the WVR will not save you from loosing the files. Restoring from a backup made with an incompatible firmware version will cause a crash, and go through a bit of a process to save it.

closed 2021-11-19 by marchingband