marchingband/wvr #30: Presets 1–16 Are Program Changes, Not MIDI Channels

TheBatmosphere, 2021-11-13:
1. Load sounds into presets 1 and 2. (The numbers at the top of the screen.)
2. Play notes on a MIDI controller on channel 1. Preset 1 sounds play.
3. Change to any other channel. Preset 1 sounds still play.
4. Send a MIDI Program Change command. Program 2 plays preset 2, program 3 plays 3, etc.
Am I misunderstanding the 16 numbers at the top of the screen? I thought they were MIDI channels, but they behave like programs. And what does "voice" mean?

[image]

marchingband, 2021-11-13:
because WVR is a sample player first, and a midi device second, it has a notion of voices. These are just the banks of sounds. It doesn't map clearly onto programs or channels in midi terms.
Currently every midi channel defaults to pointing at voice 1. program changes on any channel change what voice is pointed to for that channel, but not for the others.
If it makes more sense, I could change that to have each channel default to its respective voice, and change from there.

marchingband, 2021-11-13:
Thanks for pointing out that the `voice` field in the menu needs to be updated to the 1-16 scheme :) I'll fix that!

TheBatmosphere, 2021-11-13:
Thanks, Andrew. I had assumed the 16 buttons represented the 16 MIDI channels, so that was my misunderstanding. The concept of using programs to address sound sets is also the way the SparkFun Tsunami works. On that device, MIDI program 1 addresses sounds 1–128, program 2 addresses sounds 129–256, etc., up to sound 4,096.

However, the Tsunami also has a global option to set the MIDI channel to 1–16 or omni (all channels). That seems like a sensible approach to me. If you used 16 channels times 128 MIDI notes, you'd get 2,048 possible sounds instead of 16,384, right? (Not that anyone would use 16,000 sounds, but my point is that selecting sound sets via program changes rather than MIDI channels — which is what I assumed was intended due to the 16 slots — seems like the most musical approach.)

In short, I'd recommend adding a global MIDI channel control (1–16 or All) and using MIDI program changes to address banks of sounds. (Which I think you're doing already. TBD whether people would want more than 16 programs.)

By the way, thank you for considering all our wild feature requests, explaining why they might not work, and implementing them so quickly!!

marchingband, 2021-11-16:
Ok that helps me understand the situation a lot better.
I do still like the idea of having multiple channels mapping to their own voice.
I am imagining 2 keyboards plugged into WVR, one on USB, one on DIN, and both players able to move around between voices independently. This is how it's setup now. Each channel deafults to voice one, but can change that with program changes.
I am also seeing the benefit of choosing one channel. Often in a larger midi setup you want some devices to be only listening to events on one channel, and ignoring all the others.
Perhaps I will add a menu item to select channel 1-16 or omni. Omni would follow the current scheme, but selecting a single channel would listen only on that midi channel, and ignore the rest.

TheBatmosphere, 2021-11-20:
Yeah, I’d think you’d want each controller to transmit on a different channel so one player wouldn’t be modulating the other’s sounds. (That’s after we get MIDI controls like pitch bend and CCs. ;-) I still need to write up that feature request. 

I have a General MIDI module with simultaneous MIDI and USB host inputs, and it’s convenient. You can set both controllers to transmit on the same channel, but a more common application is to use unique channels to trigger different sounds. For example, a MIDI keyboard could play organ on channel 1 while a USB drum pad plays percussion on channel 10.

TheBatmosphere, 2021-11-21:
I just noticed that MIDI CC 125 is defined as Omni On and 124 is Omni Off. I'd prefer to set omni mode and channel in the UI, but it's interesting that it's defined.

marchingband, 2021-11-21:
Do you mean it's defined in the MIDI spec? I certainly didn't (intentionally) implement this in WVR code.

marchingband, 2021-11-21:
What you are saying about not wanting controllers on different channels to modulate each others sounds makes a ton of sense. I will keep that in mind for sure.

TheBatmosphere, 2021-11-22:
> Do you mean it's defined in the MIDI spec? 

Yes. I updated my comment above. Cheers!

stephenmaughan, 2022-02-14:
did originally think that the 1 to 16 voices were midi channels so I loaded a set of samples into
them , but when I tried to play them with different channels it obviously didn’t work. 

This is just personal preference of how I would love it to work….. each voice could have a midi channel assigned to it and you could load different instruments to each voice/channel and play each voice channel from a sequencer or DAW. Would this be possible to implement? 

I’m guessing not every one would want to work that way and I have no idea how much work that would be to get that to work with the web ui. It’s just a suggestion.

closed 2022-02-17 by marchingband

stephenmaughan, 2022-02-17:
> Closed #30 <https://github.com/marchingband/wvr/issues/30> via #36

Just gave it a quick try with the basic wvr firmware and my din socket and
it seems to work great so far. I’ll try later with the new  usb backpack
firmware.

Thanks very much for implementing it. I think my WVR is going to be turned
into a drum sample machine.

marchingband, 2022-02-18:
@stephenmaughan racks are amazing for drum hits, a kit setup that way is unbelievably realistic. So incredibly fun :)
I am down to nerd out about drum stuff anytime! Here are some pix of the kit I have been working on over the last few years.
https://hohumdrums.netlify.app/ 
I may do a kickstarter for them at some point!

stephenmaughan, 2022-02-20:
> (PS they are not actually for sale, so dont try to buy them haha, that
> website is just a prototype of an online store I was developing as a
> separate project)

They look really great. They remind me of something Kraftwerk might have
used (my Kraftwerk drum samples were one of the first I put in my WVR) or
maybe New Order.

Will you be making anymore WVR’s or will the next version be for eurorack?

stephenmaughan, 2022-02-20:
I’ve got my WVR running off battery power 3 x AAA (4.5v total) that should
ok shouldn’t it ?

marchingband, 2022-02-20:
Yes it's a 3.3v LDO, so that will work well.
Glad you like them, they were super fun to make, I don't like the look of the Roland pads so I am happy they came out looking tight :)

stephenmaughan, 2022-02-20:
The batteries seems to be working great , I’ll still keep  the usb option
though.

 If you ever do start selling your drum pads  I’m sure you’ll sell loads.

PR #36 (2022-02-17): Asr looping
overhaul the audio engine for much higher speed. There is now some extra processor time to play with, to implement more midi controls like panning, expression, etc, in a future PR :)
add ASR-looping to the engine and the WEB UI
add MIDI CHANNEL to WEB UI, and to midi engine : WVR can listen on only one midi channel, and is not stuck in OMNI mode :)
Each midi channel now maps to its respective voice at boot. ie. midi events on channel 5 play sounds from voice 5, etc. Program Change events still modify that routing, but they reset to defaults at boot.
upgrade to ESP32-ARDUINO v2.1
fi [... cut]