marchingband/wvr #31: MIDI velocity ≠ MIDI volume

solipsvs, 2021-11-18:
in a musical sampler that uses MIDI, velocity does not always equal a volume or amplitude change of the material triggered.  it is set that way in most circumstances, but i think in WVR it should not be hardwired this way.  BTW there is an actual MIDI ‘volume’ parameter that controls the output amplitude of the entire instrument associated with an instrument and all its layers and samples that is applied after mixing.  velocity should be mappable to things like sample output amplitude, velocity layer invoking (without imparting an amp change), and any other change it could be mapped to such as filter, lfo time, amplitude decay time, etc etc etc to be used in a creative way.

so currently, i am assuming ‘velocity’ is just an amp setting and is static.  this means that only one sample can reside on any note number and any variation of the sample would need to be placed on another key?  correct me if im wrong.  when you say it has velocity layers, i understood that you could place many samples on one key that were invoked according to MIDI velocity.  this is how any musician would desire to use a sample playback unit in a default way.  can this be done?

marchingband, 2021-11-18:
Yes we call these racks, they're available in the UI

closed 2021-11-18 by marchingband

solipsvs, 2021-11-18:
so are we only able to set a velocity layer from 50-127?

marchingband, 2021-11-18:
no you can set "number of layers" which refers to how many velocity layers in the rack (up to 32 layers)
then you can set the 'break-point" which is the velocity range any particular layer covers (anywhere in the 0-127 range)

marchingband, 2021-11-18:
https://github.com/marchingband/wvr#using-racks