marchingband/wvr #32: Feature Request: More MIDI Controls

TheBatmosphere, 2021-11-21:
In V1.0.1, I found that WVR supports
- velocity to level
- CC 10 to panning (kind of jumpy around the extremes)

Here's a starter list of additional controls I think would help:
- pitch bend — add a global option in the UI to set the range in semitones (0–24).
- CC 7 (volume)
- CC 11 (expression) — a percentage of volume (CC7)
- CC 64 (sustain) — 0–63 is off, 64–127 is on
- CC 72 (release time)
- CC 73 (attack time)
- CC 120 (all sound off) — instant mute (contrast to CC 123)
- CC 121 (reset all controllers)

**Pending Future Synthesis Features**
- CC 1 (modulation) — typically assignable to anything, but would be immediately useful for depth of a pitch LFO
- CC 71 (filter resonance)
- CC 74 (filter cutoff frequency)
- CC 91 (reverb amount)
- CC 92 (tremolo amount)
- CC 93 (chorus amount)
- CC 94 (detune amount)
- CC 95 (phaser amount)
- CC 123 (all notes off) — jumps to release phase of envelope
- CC 124 (omni off)
- CC 125 (omni on)

Handy summary [here...](https://anotherproducer.com/online-tools-for-musicians/midi-cc-list/)

marchingband, 2022-02-28:
I have added all of the first set, with 2 exceptions : pitch and attack.
Pitch will fit neatly into the DSP version coming soon. I may come back and try to implement it in this engine later, depending on the outcome of the DSP engine, and its polyphony capabilities.
So, I'll leave this issue open. If anyone really really wants attack, let me know, and I will write it!

robs81, 2022-04-01:
Maybe a midi SYSEX command to turn ON / OFF wifi? This can save 1 or 2 pins when using WVR with another microcontroller?

marchingband, 2022-04-02:
this is a good idea, thank you! i will implement it this weekend. ill use CC 85 i think.

robs81, 2022-04-02:
Cool! Isn't Sysex message more suited for this purpose because they don't
depend on midi channel. Cc are midi channel related.

marchingband, 2022-04-03:
True! I added sysex controls for WiFi on/off.
0x01 is WiFi On
0x02 is WiFi Off
so the sysex would look like "F0 01 F7" for WiFi-on and "F0 02 F7" for WiFi-off.

marchingband, 2022-04-03:
https://github.com/marchingband/wvr_binaries/tree/main/wvr_basic/3.2.0
and all the firmwares at v3.2.0 have this feature.

TheBatmosphere, 2023-06-06:
Hello again. Any more thoughts on adding pitch bend? I wonder if it could be implemented by adjusting the sample playback rate rather than via fancier DSP. That would probably create all kinds of gritty artifacts, but might sound cool.

marchingband, 2023-06-06:
Crude pitch-down can be done by doubling every n sample, but it sounds bad. Crude pitching up can be done by skipping every n sample, but then it requires reading more samples from eMMC and using larger buffers, so that's not possible without sacrificing voices of polyphony.

I have been looking at DSP using fixed-point math, for which there may be enough time in the processor to keep a decent number of voices of polyphony, but it will be a lot of work, and I may not get time for it soon. I am not sure if it will be worth it, if we end up with 1/2 as many voices, which is likely for 1 octave of pitch up.

The only easy option is low-quality pitch down, I could implement that for fun maybe within a weekend, and create a special binary with it enabled but it's not a high priority at the moment.

TheBatmosphere, 2023-06-07:
Thank you for thinking about this. Most synths default to just two semitones up and down. An octave down could be cool for occasional special effects. 

The render-and-crossfade approach strikes me as overly complicated, but I’m not a programmer. 😜 As a player, I’d be fine if the timbre changed radically as the pitch bent; it would give the instrument more character.

CarlosUnchained, 2023-12-10:
Regarding WIFI on/off via midi, in my case would be easier to have a CC to toggle it additionally to sysex.
Just make it so it can be modified by any midi channel.

marchingband, 2024-01-03:
added pitch-bend in 3.9.0