marchingband/wvr #37: Feature Request : play sample with start and end parameters

marchingband, 2022-02-20:
WVR class should have a method `::play(uint8_t voice, uint8_t note, uint8_t velocity, size_t from, size_t to)`
The sample will start playback from sample index `from` and stop at index `to`
The sample will need to be faded-in if `from` is not equal to 0.
The sample will need to be faded out if `to` is not equal to 0 or `sample.len`

marchingband, 2022-02-20:
`::play(uint8_t voice, uint8_t note, uint8_t velocity)` will just call `::play(voice, note, velocity, 0, 0)`

marchingband, 2022-02-20:
write 2 examples as described here : https://groups.google.com/g/wvr-audio/c/7Rdht_m435M/m/l63Cs6e4AgAJ

CarlosUnchained, 2023-10-25:
Hello, I’m new here. Any progress in this topic? I’d love to have CC control over start point of the sample.

marchingband, 2023-10-25:
No progress yet, I havn't been able to find time to add new features in quite a while, sorry.

marchingband, 2023-10-25:
I am not even sure it will be possible to build this feature to be honest.

CarlosUnchained, 2023-10-25:
Oh, that's a pity. I understand tho, thanks for the answer!