marchingband/wvr #49: Possible Bug: ASR Loop Ignores Note-Off Halt Setting

TheBatmosphere, 2022-07-18:
With a loop in the middle of an audio file and note-off set to halt, holding the key causes the sound to loop as expected and then play to the end when the key is released. However, if I release the key _before_ the loop start point, the sound doesn't halt. It plays like a one-shot, all the way to the end. I'd expect playback to halt.

P.S. — ASR looping didn't work at first, so there may be another glitch in its implementation. I'm not sure what I changed to get it to start working. Cool feature, though!

[image]

marchingband, 2022-07-18:
Glad you are liking the ASR loop mode :)
What you describe is the intended behaviour! It's modelling an envelope, which always plays the release (R). So we have to also always play the Sustain section to avoid pops. I could program it differently, as you said you expected, but it would cause problems for others, for example if the Release section is a reverb tail. It's an quirky and imperfect analog to the synthesis ADSR equivalent. It is what it is!

If you need the behaviour you expected, I could look into how that would be implemented, and possibly create a custom version for you.

TheBatmosphere, 2022-07-18:
Thanks for the quick reply. Isn’t that case handled by setting note-off to “ignore”? In other words, with note-off set to “halt,” the sound would quickly fade when the note is released, no matter where it was in playback. With note-off set to “ignore,” the entire sound would play. I think that would make ASR loops respond the same way as full-file loops.

marchingband, 2022-07-18:
In both loop and asr-loop mode, 'note-off ignore' means the sound should loop forever, because it never receives a note-off event. Not something many people would want, but just the natural outcome of the logic.

The behaviour you expect would require a 3rd option, where note-off events are treated differently depending on when they occur .. before or after the loop-start. Before would cause a fade-out, after would finish playback. It sounds useful, but it would require a new setting to keep the logic consistent.

TheBatmosphere, 2022-07-18:
Hmm. Said another way, what is the use case for having a loop ignore the note-off? Seems like “ignore” would be useful only for one-shots.

marchingband, 2022-07-19:
Ok I see now. So I could change it from halt/ignore to halt/play-through or something. That makes sense and I think covers all the cases. In the ASR case, when using halt, I would expect it to fade-out whenever note-off is received, so we would never hear the release at all. But you're suggesting in this case it only halts if it's in the Attack section? That seems inconsistent doesn't it?

TheBatmosphere, 2022-07-19:
Yes, I was thinking the sound should fade out as soon as the note-off is received, just like in the other loop mode. Looking forward to ping-pong looping too! Thanks for all you do.

marchingband, 2022-07-20:
Ok cool. I'm a little preoccupied with some commission work at the moment but I'll make some time ASAP to add this. Thank you for continuing to contribute ideas and feedback, I really appreciate it!

marchingband, 2022-09-22:
implemented in 3.3.0

closed 2022-09-22 by marchingband