marchingband/wvr #51: Feature request: MIDI over BLE

vamix2012, 2022-09-01:
I have seen lots of BLE Midi projects, my accordeon also uses BLE Midi at this point, it would be cool if we can connect to the WVR via BLE, I have seen that every BLE capable microcontroller can send and recive MIDI messages, check this link :  

https://github.com/lathoub/Arduino-BLE-MIDI/blob/master/examples/MidiBle_Client/MidiBle_Client.ino

And this guy shows how two esp32's are connected together via BLE: 

https://youtu.be/UgI7WRr5cgE

Please tell me if that can be possible?

Thank you very much!

marchingband, 2022-09-01:
This is awesome! Yes I'm sure it can be done. I'll make it a priority to try ASAP.

marchingband, 2022-10-19:
oh for crying out loud the BT stack is GIGANTIC, it adds almost 400kb to the binary size, WVR would need to have it's flash partitions changed in order to fit the drivers. This is a feature that I think will only be possible to add via Arduino IDE