marchingband/wvr #56: Support Decent Sampler Format

maks, 2022-10-07:
Along the same lines of #55 another quite nice sampler format is [Decent Sampler by David Hilowitz](https://www.decentsamples.com/) and it has quite a few sample libraries for it by him and others.

The format is XML based and [well documented](https://www.decentsamples.com/wp-content/uploads/2020/06/format-documentation.html) and while I would expect it pretty complicated to implement the whole spec, a minimal implementation is fairly easy and I did one in [TS for my ML-1 project last year](https://github.com/maks/ml-1/blob/main/src/sampler/dspreset_parser.ts).

I guess the first step to this would be finding a nice, lightweight C/C++ xml parser lib.

marchingband, 2022-10-07:
Hey Maks!
I will look into this!
FYI preparation for the samples is all currently done in JS in the browser before upload to WVR. WVR actually only modulates volume and pan in real-time, the rest is pre-rendered destructively in the browser.
The goal for #55 is to parse loop points, start/end points and velocity layers, although it wouldn't be too hard to add FX already setup in the WVR GUI, and for that matter, any FX that the Web Audio API implements.

Is this more or less what you had in mind here as well?

marchingband, 2022-10-07:
Here is the repo for WVR GUI https://github.com/marchingband/wvr_ui

maks, 2022-10-07:
Brill thanks Andrew. 
Sorry I'm only starting to get look around and get the hang of how WVR works. I should have realised from the docs Readme about reverb being applied in-browser that this is how it worked, makes sense!

So yes, I think that would make it even easier to support DS format as then its "just" a matter of implementing it browser-side so I really need to get around to pulling out my existing code into a proper npm module for reuse.

marchingband, 2022-10-07:
Looking more closely at this, I think a decent subset of the `<group/>` parameters will jive quite nicely with WVR code.
The `loopCrossfade` parameter is really cool, but that would be a bit of labour to implement.
The `releaseTrigger` parameter is something I hadn't considered, I am assuming it allows you to specify a release sample to be played as a part of another main sample? That's brilliant, and I am annoyed that I didn't invent that.
The `loNote` and `hiNote` parameters are very cool. Is the idea here that you would pitch shift a given sample to fill up a range, but only so far, above/below which another (closer) sample would be pitch shifted to pitch? I guess that way you could get a very large range of notes, with a small number of samples, and without the need to pitch shift too far and lose quality. Also mad I didn't invent this.
The legato system is mind blowing. Is the idea here that you have a special legato sample, which you pitch shift into place so that any 2 adjacent notes can be legatoed!? Amazing, and I am shocked that this approach works.
Lots here to learn from, thank you for bringing this to my attention!

marchingband, 2022-10-07:
@maks your ML-1 project is MIND BLOWING. I am speechless.

maks, 2022-10-07:
@marchingband oh thanks, I'd actually want to say the same about WVR! the more I'm reading through here, the more I'm so impressed with all the functionality you've been able to pack into it, u wish I had come across it earlier!
With ML-1, I got it to this point: https://youtu.be/ZcfssfH9EQY but then kind of ran out of steam, realising how much work it was going to be implementing all the features I wanted. So I've actually started ML-2 recently which is also using the AkaiFire and running on a RPI4 CM but using the audio/sequencing engine from Sunvox, which is available as a C lib for Linux (as well as android, iOS, macos, Qin & Wasm). At least then I only need to implement only the "hardware UI" and Midi and lean on Sunvox for audio part. The code for it is still bit of a mess, but I'm planning to publish it soon. It's done in Dart using Darts FFI for bindings to all the C libs I use, so in theory it could work in a webbrowser in future, but for now I'm just targeting running in a Rpi with a minimal Linux build.

But I'd like to experiment with playing back very long samples and also FM synths, so that's my plan to use WVR and WVR being able to be a usbmidi host is great for using with the Akai too, though I think I'd need to do custom firmware to light pads & drive the Fire's OLED (it's all controlled via MIDI sysex).

Getting back to the topic of Decent Sampler I'm afraid I don't know the format well enough to answer all your questions, but I think I can on some.
With the hi and low ranges, yes the idea is that you have samples to cover ranges of notes so you don't need to do much pitch shifting and the there's also the facility to specify round robins and a bunch of other things as you've seen. I have to admit for ML-1 I only implemented the bare minimum of support I needed to use the sample packs Loopop made for contest entries but I'd like to implement more once I get my WVRs 🙂

marchingband, 2022-10-07:
I may have lost steam too, but having Kickstarter backers waiting on their stuff gave me some much needed accountability :)

Adding MIDI output to the XIAO would take a few lines of extra code on the XIAO, and a jumper to add UART TX connection, currently there is only RX. I have considered writing something similar to target launchpad LEDs, but this Fire looks even better.

Dart!? You are really a polyglot! There is something like Sunvox called Faust which compiles for ESP32, I have done some tests that have worked well, but I don't know how much DSP an MCU can really handle. It may be on my winter todo list.

Excited for you to get your boards and start testing!

I am pleased to see from your parser that XML is so straight forward to read in JS, have never tried that. It would probably just be an afternoons work to get the WVR UI to read a file with the basic parameters you have set up in your parser.

Gnuv, 2022-10-24:
Bjoerns Sample Mapper can translate sfz to decent sampler format. I didn't try yet, but maybe it could be suffisant for creating both sfz, decent and WVR instruments :
https://www.bjoernbojahr.de/bjoerns-sample-mapper.html