marchingband/wvr #59: Feature: Add attack and release controls

marchingband, 2023-05-29:
Attack and Release will default to 0ms
CC72 will be release, the value expressed in units of 50ms, so 0~127 = 0~6.35 seconds
CC73 will control attack, 0~6.35 seconds
CC22 will become the instant fade-out time, with the same units as it currently has.

The curve of the attack and release will be inherited from a given sounds velocity curve.

TheBatmosphere, 2023-05-30:
Remind me again how CC22 would work and how CC72 would interact with it? I'm still not clear why it’s necessary to adjust the instant fade-out time. (It wouldn't be instant after adjusting, right? ;-)

Linking the fade curve to the velocity curve is an interesting idea. I wonder if that would cause unexpected behavior. 

I was originally thinking of having CC22 switch among several curves, for example:
- 0–42: concave exponential (curve bends down)
- 43–85: linear
- 86–127: convex exponential (curve bends up)

But I suspect a single curve would be enough. Many ADSR diagrams show convex attack and concave release. That supposedly reflects [how we perceive loudness](https://www.earlevel.com/main/2013/06/01/envelope-generators/). (Also see "response curve" in [this article](https://www.perfectcircuit.com/signal/learning-synthesis-envelopes-1).)

[image]
(Image from [AAS](https://www.applied-acoustics.com/multiphonics-cv-1/manual/adsr/))

TheBatmosphere, 2023-06-06:
Thanks, Andrew. I still can't imagine when I'd need to use CC22, but having attack and release would be great. I'm curious to play with the curve shape too — that may be a first on a hardware sample player.

marchingband, 2023-06-06:
I can't remember what kind of sounds required a slower ASAP fade-out, to avoid the pops, it was a while ago, but I ran tests and confirmed that it was indeed needed for some sub-set of sounds.

TheBatmosphere, 2023-08-14:
FYI, I was playing WVR with attack and release knobs last weekend and found them super-useful. Thanks again. I’m not yet sure about the curve.

marchingband, 2023-08-14:
Good to hear!
Once we decide on the curve I will probably add this to the main firmware.

robs81, 2023-10-13:
Attack and release up to 6s would be great... Will this be added to main basic firmware?

marchingband, 2023-10-13:
I have pitch-bend, envelopes, WiFi station mode, and a few other features ready to merge into the firmware, they just need rigorous testing, which I havn't had time to do quite yet.

Next chance I get I will make a new "release candidate" firmware and send out a note to request testing of it. 

I'll post that in the Google group and in these GH issues as well.

marchingband, 2023-10-14:
@TheBatmosphere do you agree with me that inverse-square-root is the only curve that seems useful?

TheBatmosphere, 2023-10-15:
Thanks for asking. I’ve been using linear. I’ll go back and compare the others ASAP.

TheBatmosphere, 2023-10-23:
I tested attack and release curves by loading the same sample into adjacent notes and setting their velocity response to linear, square root, and inv square root. Inv square root definitely sounded the smoothest. (I tried the fixed response too, but suspect it's using the inv square root curve because it sounded the same.) 

One weird thing is that velocity seems to affect the fade duration. Maybe that was just for testing, but I think it would sound more natural if it were decoupled. In other words, velocity would affect just volume and CC 73 and 72 would affect attack and decay with an inverse square root curve. Thanks for considering this feature!

marchingband, 2024-01-03:
done in 3.9.0

closed 2024-01-03 by marchingband