marchingband/wvr #61: Feature Idea: Play/Pause/Resume

TheBatmosphere, 2023-08-14:
Not sure how this could be implemented in WVR, but the SparkFun Tsunami has a handy play/pause/resume feature I use all the time for long speech samples. One trigger starts a sample from the beginning, another pauses playback, and a third resumes it. That lets me perform phrases to the situation without needing to assign each one to a unique note. 

In my configuration, the pause button stops playback of all samples, even ones not triggered by the play button. (Like WVR, Tsunami is polyphonic.)

Perhaps this could be implemented on WVR with pin assignments or even a MIDI control change. If CC value is 1–63, retriggering notes would start their samples from the beginning. If it’s 64–128, playback would resume from the current position. 

Come to think of it, maybe this would work better as a new trigger mode.

marchingband, 2023-08-14:
Interesting ... yes maybe note-off:pause as a new option, then midi and pin events alike would trigger the same behaviour. I'm not sure it would be easy, but it might be.

TheBatmosphere, 2023-08-14:
Cool! I wonder if it would need a reset command too, for when you want to retrigger the sample from the beginning. Maybe CC120 (sound off) would handle that. 

https://github.com/marchingband/wvr/issues/32#issue-1059310153

TheBatmosphere, 2023-09-24:
Another thought: Instead of using a CC to reset the playback start position, maybe we could designate another key/pin to be a global reset. For example, C3 could start and stop the sample and C8 could reset the start point. Or...when trigger mode is set to Play/Pause, hitting another key in the same exclusive group would reset the start point.

marchingband, 2023-09-25:
It seems to me the easiest way would be simply a second note that has the same sample, but with the normal note-off configuration ... would that solve it?

TheBatmosphere, 2023-09-25:
Well (if I understand correctly), having a second copy set to Restart mode *would* allow the performer to start the sample from the beginning. But what I was thinking about was how to deal with leftover audio in the original note. You’d have to play it all the way to the end to use that key again. 

I’m imagining a museum installation or audio tour where sometimes you’d want to pause and resume a long speech and sometimes you’d want to restart. But maybe the CC would be a better way to handle that case. It is awkward that the Stop command on the Tsunami stops *all* audio.

TheBatmosphere, 2023-09-25:
Thanks for letting me think out loud. :-D 
I think CC120 could do it. For installations, perhaps a pin could be configured as All Notes Off as well.

marchingband, 2023-09-25:
It's very helpful thank you! I tend to focus on my own use cases :)
Yes pin function option for CC120 as well. I imagine the pins would be the easiest way to trigger sounds in an installation situation.  
I believe that the combination of 
```
retrigger: note-off
note-off: pause/resume
```
would allow an installation to have press a single start/pause button to start/pause, and another button to reset, which seems complete.  
Having said that I also see the logic with the tsunami decision to have start, pause, and reset all on separate buttons, and therefor both pause and reset acting on all playing samples.

TheBatmosphere, 2023-09-30:
I updated my comment [above](https://github.com/marchingband/wvr/issues/61#issuecomment-1732782442) after rechecking my Tsunami configuration. Triggers in Tsunami operate on ranges of notes. I had my Stop trigger set to stop all notes, but what I should do is wire up a second button to control a subset. The new modes make sense, though I wonder if you could consolidate them to retrigger: pause/resume.
```
retrigger: note-off
note-off: pause/resume
```

TheBatmosphere, 2023-10-23:
Conceptually, the new retrigger mode makes the most sense to me. In both cases, the player could use a note in the exclusion group to reset the playback position to zero.

marchingband, 2024-01-03:
done in 3.9.0

closed 2024-01-03 by marchingband