marchingband/wvr #63: Crashing esp32

drfhaust, 2023-09-06:
Hey I uploaded the code to an esp32 without psram. It has 2 outputs. 1 where it crashes because of a flash error and another there I disconnect the sd card  and it gives this 
`Rebooting...
Total heap: 306416
Free heap: 278852
Total PSRAM: 0
Free PSRAM: 0
CPU freq: 240
after begin, remaining: 232412, used:4294833488
Total heap: 306416
Free heap: 278852
Total PSRAM: 0
Free PSRAM: 0
CPU freq: 240
after begin, remaining: 232412, used:4294833488
E (37) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
Name: 
Type: MMC
Speed: 0 kHz
Size: 0MB
CSD: ver=0, sector_size=0, capacity=0 read_bl_len=0
after emmc, remaining: 231604, used:808
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d37dd  PS      : 0x00060c30  A0      : 0x800d8c45  A1      : 0x3ffb2770  
A2      : 0x00000000  A3      : 0x3ffc426c  A4      : 0x3ffc4268  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x0000e100  A8      : 0x00000000  A9      : 0x3ffb2750  
A10     : 0x00000000  A11     : 0xdd6bb9fb  A12     : 0xdd6bb9fb  A13     : 0x3ffb270c  
A14     : 0x3ffc49b0  A15     : 0x3ffb270c  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008abd9  LEND    : 0x4008abe9  LCOUNT  : 0xfffffffe  

Backtrace: 0x400d37da:0x3ffb2770 0x400d8c42:0x3ffb2790 0x400d55f0:0x3ffb27e0 0x400d2f2c:0x3ffb2800 0x400e412a:0x3ffb2820

ELF file SHA256: 93d89fea29b2fb86`

marchingband, 2023-09-06:
Hello. This codebase is for a specific hardware device called WVR, not just any old ESP32. It expects to find PSRAM, an eMMC module, a DAC, etc, and will definitely crash without it.

drfhaust, 2023-09-06:
Hi, can I swap put the sd card for the emmc, they are the same pinout, also i plan on swapping out the dac with a dsp(the adau1701) does your cide have any functions to detect if it's not the pcm5102. 
I'm sorry if I'm being forward. I am designing a device that plays wavs in banks using 6 buttons. It also sums the wavs with a microphone  and passes the signals to a recorder.
I have this working using an stm32 and an atmega328 but I am trying to lower the bom, that was when I saw your awesome device.
Can I ask some questions? What is the main use of the psram? Caching or for the network functions
[image]

marchingband, 2023-09-06:
You will have to modify the code a bit to use SD card,  and to use a
different DAC.

SD is slower then eMMC so you will not get 18 voices of polyphony. You
could get 4 or 5 possibly.

All of the real ram is used for the audio player, so the PSRAM is used for
everything else, you will see ps_malloc throughout the code which is for
PSRAM.

You can ask questions, but I am on vacation for the next 2 weeks and will
be slow to respond. I am not familiar with that codec, and I am not sure
how much my code will help you to be honest, but you are more then welcome
to try!

drfhaust, 2023-09-06:
Thanks fir the responsr. Hows your vacation going.
Please Correct me if I am wrong, but since the esp32 is just the i2smaster, I was of the opinion that it would basically spit out i2s data. I was not aware there was any communication between the pcm5102 and the esp32 asides from the audio data.
That was why I assumed that The adau, which  basically has a configurable i2s would simply connect to that port and it would work fine

marchingband, 2023-09-06:
I am only 1/2 way to my destination :) I believe you are right. PCM510x has its own clock, so I do not send a clock. Other then that it's likely the same kind of i2s data.