marchingband/wvr #66: Feature request: Reverse playback via CC control

CarlosUnchained, 2023-10-25:
It’d be nice to have an option to play samples in reverse before they are triggered. My guess is that a CC number with value over 64 could enable the reverse playback.

marchingband, 2023-10-25:
i don't think that will be possible on the playback end. However I could add it to the WEB UI ... so you would have 2 notes, one forward, one backward ... would that work?

CarlosUnchained, 2023-10-25:
My use would be standalone with the eurorack shield, so not really. This and the start position would unlock a ton of sonic possibilities but I understand the limitations and the scope of the project. All good :)

marchingband, 2023-10-25:
Ok that makes sense. On closer inspection I don't think reverse and play-from-a-to-b would be so hard. I'll put these in the ASAP pile.

marchingband, 2023-10-25:
In the meantime I think reversing the sample in a DAW and saving it to a second voice would work for what you're describing. Maybe voice 1 is all the samples forward and voice 2 is all the same samples backward, then you use 2 channels in midi to pick one or the other, or send program change commands around the ones you want reversed to set that channel to voice 2 before the note, and then back to 1 after.

CarlosUnchained, 2023-10-25:
Oh woah, that's great to read! I had already given up on these two features hah.
I'm still trying to get an answer from the maker of the shield to get the thing itself so no rush. Independently of the features above, I think it's a killer project and I'll get it as long as he is willing to sell one.

Oh that makes sense. I was unsure of what PrCh actually does, only read that it can receive it. So it sets the midi channel per voice? I was wondering how to talk to each voice to control CCs independently. I still have some knowledge gaps on midi implementation regarding multi samples. Only saw midi channel mentioned on global settings.

marchingband, 2023-12-10:
Sorry, I missed this before.
PrCh has an unusual implementation in WVR, it controls what voice a given channel is set to.
By default each channel is set to the corresponding voice (ch1 is voice1, ch2 is voice2, etc).
So using PhCh, any channel can access any voice. 2 or more channels can also use the same voice.
For WVR, a "voice" refers to a full set of samples, and this is how the UI is setup, there is a list of voices near the top of the screen. You can toggle through and upload samples to each note, for each voice. If you are in "rack" mode (multisample) then you can upload a sample for each layer, for each note, for each voice.
Maybe you figured this out already now that you have the module in hand.

CarlosUnchained, 2023-12-10:
Thanks! I figured it out yes. 
I have a limitation (not a big deal) with this and it’s that the digital inputs are linked to midi channel 1, so in order to access other voices with them I have to change channel 1 to another voice and the corresponding voice channel to 1, so it’s a bit confusing. 

Just saying this is case there’s another way of changing which voice the D pins trigger independently of midi channel. 

Thanks again!!

marchingband, 2023-12-10:
There isn't another way ... it could be fairly easy to have another drop-down menu in the pin config page, to select the midi channel for events on a given pin. Would that really help your situation? I am not sure I see the issue clearly

CarlosUnchained, 2023-12-10:
TLDR: feature request:
- unlink D pins target voice selection from midi channel 1
- select D pins target voice with a midi CC# (values 1-16, cap above) sent from any midi channel

It's not an issue really, just inconvenience. Let me explain again just in case:
Say I want to use (in eurorack shield case) the buttons and gate inputs for voice 10, if I have also midi controlling voice 1 on channel 1, I have to do the trick above, that is sending prch 9 on channel 1, and prch 0 on channel 10, then also exchange midi channels so the voice 10 controller has to be set to channel 1 and the voice 1 controller to channel 10.

Surely this won't be needed often if planed ahead. Additionally, it allows to move what voice is affected by the digital pins without the web app, with the drop-down menu you mention looks it'd be fixed.

However, what could be an improvement from my point of view (feature request) is having a midi channel agnostic way of changing the voice all the digital pins trigger. For example using a fixed CC number independently of channel number, which value sets the voice - D pins relation, from 1 to 16 and cap above values to 16. From the eurorack standpoint, changing this CC is like changing for example a drum kit dynamically, this becomes even more interesting by having some kind of automation on that CC, since the D pins would be triggering samples through different voices in a single pattern. Note off issues that may arise can be prevented with the current trigger configuration.

This would give more freedom to midiless playback (I have a voltage to midi converter on the rack so I could set a knob to change a CC value via TRS and thus the target voice of the pins). Fun times!

I do not see useful the need to have different pins on different voices as you can always make a specific voice covering all your needs and it's easy to get lost, but to be able to change target voice without affecting midi channels as proposed above, is definitely an improvement.

Sorry for the long post, I wanted to be as clear as possible. Thanks for reading!

marchingband, 2023-12-12:
I think the midi-style solution to your situation would be:
set your midi controller to send on channel 2.
pgch sent on channel 1 now controls the gate-controlled voices.
pgch sent on channel 2 now controls the midi-controller voices.

It sounds like you want them to both start on voice 1, so you would start by sending pgch 1 from your midi-controller. Now both are using voice 1, but that can be changed independently. Am I understanding correctly?

Midi channels are typically used for only one controller each.

marchingband, 2023-12-12:
Hmmmm. 

Could you duplicate voice 1 to a second channel to solve this?

CarlosUnchained, 2023-12-13:
You are right it's not the cleanest solution. Also what you propose it's a workaround but it's good enough, yeah. I was focusing on using voice 1. While not ideal, "loosing" it it's not a big deal.

I think a good alternative would be to not use MIDI but the Dpin inputs to change voice target for the rest of the pins, without affecting MIDI at all like explained above.

Regarding this, I see there's a bank up/down on the pin settings, I have not tested that (don't have time at the moment).
What does it change?
If it's to change voice target for all the pins that's just what I wanted and has been there from the beginning :)