marchingband/wvr #67: Feature request: note off - toggle

CarlosUnchained, 2023-12-09:
Hi!

Finally got WVR eurorack shield and loving it so far. I have a request that benefits any playback workflow on WVR and that's toggle sample. This means that a midi note (on/off) starts the sample, and the same note (on/off) stops it.
On the eurorack shield you can only play long samples by holding a button or having a gate, which is not as convenient as pressing once to play and again to stop.

This would be also useful if you have a midi pad controller, in which a note would start a sample and another press to stop it instead of having to hold it with a finger.

I suggest adding it on note off settings, since mode would be still relevant.

PS: The google group is a spam mess.
Cheers!

marchingband, 2023-12-09:
Hello,
This feature is available, it is `retrigger`: `note-off`
https://github.com/marchingband/wvr#understanding-retrigger-mode

CarlosUnchained, 2023-12-10:
Oh thanks!
There's one situation that doesn't work, and that's with loop or adr loop selected.
In that situation note off stops the playback even if it's set to ignore.

[image]

marchingband, 2023-12-10:
oh man, ok, yes this is bug!
I am planning to get a release out around new years, I have some time over the holiday.
Thank you!

closed 2024-01-03 by marchingband