marchingband/wvr #70: Feature request: Exchange voices is the web app

CarlosUnchained, 2023-12-13:
It would be nice to have an option to exchange voices in the web app. This would help to re organize voices without having to delete and re upload the content for each voice. I say exchange, because there's no data loss in that process, so for example exchange voice 5 samples and settings for voice 9.

marchingband, 2023-12-13:
That would be great but, the WEB UI doesn't have all those files. It would have to fetch them from the WVR, and then re-upload them to their new voice. There are a number of convenience features like this that arnt possible because of this.

marchingband, 2023-12-13:
What I have considered in the past is a big matrix to "patch" channels to default voices, which could fulfil the same function. Would that work?

CarlosUnchained, 2023-12-14:
Ah I see. Makes sense. The matrix would help visually to do what you can with PrCh. I'm def not a fan of having different channels assigned a voice other than the same channel, with that many voices is easy to loose track. The matrix could help in that regard by having visual feedback when needed.

On another topic, don't want to start an issue for a question: How is the mixing approached?
I think it's pretty good for the amount of sources and levels to handle, I'm not sure if there's room for improvement.
I run into clipping with some multi sampled instruments. Gain staging is a bit hard and maybe some kind of limiter would be nice to have, also simple compression. Again, no idea what's in there already and if there's any room left.

I have been recommending the project on different sites and people seem to be interested :)
Thanks again for all the feedback so far!

marchingband, 2023-12-14:
Also FYI I created a discord server, using the new community mode, to replace the Google group. It's a handier place to ask questions, if you like. I am still getting the hang of it, but this invite should work. https://discord.gg/WugqSsUXTf

marchingband, 2023-12-14:
Oh and thank you so much for spreading the word on WVR! I appreciate that so much, I would love to see the community grow more this year :)

CarlosUnchained, 2023-12-14:
> Is it the case that you are using all 127 notes and require more voices?

In my case each voice is an instrument, and I may want to control them from different sources, specially with button/gates of the Eurorack shield. Think of a voice as drum racks, pools of textures or multi sampled chromatic instruments. Mentally is way easier to have instrument per voice, and not several split by note ranges.

I can't transpose what the gates and buttons target without pin reconfiguration so the obvious is to change bank (channel1).

Thanks for the tip regarding levels. I'll try a different approach for the samples I've had troubles with. Also saw there's a way to adjust overall volume with the pins, may try that too.

Great to see there's a Discord! Just joined. I'd be great if there could be a chat-like board to talk or share related content, I have a little video using WVR that may help you to see how I use it with midi. It doesn't really deserve its own post.

marchingband, 2023-12-14:
Thy makes sense. I havn't spent a lot of energy considering the combination of pin triggered notes and midi triggered notes. The eurorack module does present some new use cases. Let me think about it.

marchingband, 2023-12-14:
Added a general-chat text thread to the discord. Thanks for the suggestion!