marchingband/wvr #77: Global volume goes 0

kim-Limfjordsteatret, 2025-02-03:
Hi 

Don't know if it's an issue, or supposed to, but the global volume keeps resetting to 0 when I turn off the wvr board.
If it's supposed to do that, is there a easy way to turn up the volume, without having to log in every time?

Best regards
Kim

marchingband, 2025-02-03:
Hi @kim-Limfjordsteatret that sounds like it could be a bug. 

The process to modify the global volume is:
1) set the global volume in the WEB UI (it is the first setting in the "WVR" menu tab)
2) click sync

Is this what you are doing?

If it is a bug, then one way to fix it is to reset the EMMC (there is a button in the UI to do this).
You will lose all your sounds etc, and get a factory reset, but if it is a bug, this will fix whatever is going on.

If you would like to try to debug the issue further with me, let me know if you have any idea what may have triggered this bug.

It is very rare to encounter this kind of bug, this may be the first time.

Thank you.

kim-Limfjordsteatret, 2025-02-18:
Hello @marchingband 

Sorry for my long absence.. I caught the man flu - barely survived - But now I'm back ;p

Yes I modify the global volume by setting it in WEB UI and then clicking sync - It says sync complete, at all works fine until I turn off the power, then it goes 0 again... 

I'll try the eMMC reset, and see if that helps.

marchingband, 2025-02-18:
I am not able to reproduce your issue.
my steps:

PINS MENU:
set d5, d6 to volume up, down, edge falling

WVR MENU
fiddle with global volume and reset

Everything works as it should.

Can you give me more debug information?
- are you setting `edge`:`falling` in the PINS menu for the pins that are controlling volume?
- which pins are you using?
- do you have a USB backpack installed?
- are you running the firmware that the WVR came with? Or is it custom? If custom please paste it here.
- WVR should always have volume 127 when it it fresh in the mail ... was yours 0!?
- where did you buy it from, and when?

kim-Limfjordsteatret, 2025-02-19:
Hi

I'm not using pins for controlling volume - I'm only using pin D1 for triggering the sound. (and thats set to edge:falling)
I'm not sure I know what you mean by USB Backpack - Its getting power from a usb walladapter 
I have downloaded the latest firmware from your github.
The volume was 0 after the eMMC reset, and goes to 0 when power has been off.
I bought it some time ago (1.5 year) from Let-elektronik.dk  - I kinda forgot about it, until a project came up not long ago, that was perfect for the board, so its been unused until now.

marchingband, 2025-02-19:
Ok that helps thank you.
Can you paste the URL for the exact firmware you uploaded? I should be able to fix this once I know that.
Thanks for your patience.

kim-Limfjordsteatret, 2025-02-20:
Hi 
I'm using this firmware: https://github.com/marchingband/wvr_binaries/tree/main/wvr_dev_board/3.9.0
I hope that helps you
Thanks

kim-Limfjordsteatret, 2025-02-24:
That makes sense - When i bought the wvr it was called WVR developement board, so I assumed it was the right firmware. 
I have tried the new/other firmware, and for now it seems to be working. I'll keep testing, and let you know if some issues occurs.

Thank you for the help.

closed 2025-06-14 by marchingband